JUANIMAN/PerfMTK #5: System UI bug

Hinhnek, 2024-11-16:
Devices: Samsung Galaxy A03S
Android version: 13
Running this module on mtk device

After installing PerfMTK module then restart my devices, after unlock it cannot go to launcher, blackscreen and lockscreen appear again, launcher problem? No i just install another launcher(Lawnchair) for Play Store then this happen again, i install by clicking to setting button in notification panel for open app manager then open playstore for install. The devices normally show "Android is starting" but after install the module it disappear. Ok the performance after install is better but i get that bug then fixed after uninstall the module one.Please fix

JUANIMAN, 2024-11-24:
Your Samsung Galaxy A03S has Mediatek MT6765 Helio P35 with PowerVR GE8320 GPU, you are lucky you don't have a bootloop, this module is only for Mediatek with Mali GPU

Maybe it's the system.prop configurations, I leave you a version of the module without them
[PerfMTKv6.9_LITE.zip](https://github.com/user-attachments/files/17891382/PerfMTKv6.9_LITE.zip)

JUANIMAN, 2024-12-05:
> Devices: Samsung Galaxy A03S Android version: 13 Running this module on mtk device
>
> After installing PerfMTK module then restart my devices, after unlock it cannot go to launcher, blackscreen and [...]

In the latest version of the module you can choose if you want these settings during the installation

closed 2024-12-10 by JUANIMAN