JUANIMAN/PerfMTK #7: How does this module change GPU frequency/governor?

IsHacker003, 2024-11-30:
On Mediatek devices, you can't change your GPU frequency and governor with apps like kernel 
adiutor, performance tweaker, etc unlike Qualcomm devices. So how does this module do that? I would just like to know, because I just want to change my GPU frequency freely and no other tweaks are needed.

JUANIMAN, 2024-12-01:
It is only necessary to correctly assign the owner and group of the script, in my scripts the owner is root (0) and the group is shell (2000)

IsHacker003, 2024-12-03:
> It is only necessary to correctly assign the owner and group of the script, in my scripts the owner is root (0) and the group is shell (2000)

Thanks for the reply, can you please give more details? Like I want to adjust the frequency of the GPU just by running commands in termux, without installing the module.

JUANIMAN, 2024-12-03:
To apply the commands from termux you have to move to a suitable directory such as: /system/bin
`su`
`cd /system/bin`

The main scripts of my module are in its /system/bin directory, I will soon release the source code of these

closed 2024-12-10 by JUANIMAN

NCLnclNCL, 2025-06-07:
> > Thanks for the reply, can you please give more details? Like I want to adjust the frequency of the GPU just by running commands in termux, without installing the module. I also checked all the .sh files inside the module but wasn't able to find anything that affects GPU frequency?
> 
> To apply the commands from termux you have to move to a suitable directory such as: /system/bin `su` `cd /system/bin`
> 
> The main scripts of my module are in its /system/bin directory, I will soon release the source code of these

How to underclock?

NCLnclNCL, 2025-06-07:
> On Mediatek devices, you can't change your GPU frequency and governor with apps like kernel adiutor, performance tweaker, etc unlike Qualcomm devices. So how does this module do that? I would just like to know, because I just want to change my GPU frequency freely and no other tweaks are needed.

How to change