JUANIMAN/PerfMTK #8: Higher performance in power save profile

dbar24, 2024-12-05:
My device is a ULEFONE Armor 24 running magisk 27 and Android 14. I am also running the latest 7.5 update. My PS2 emulator runs better on the power save profile. The balanced and performance profiles result in the emulator running with lower FPS and at a lower speed. The emulator settings are unchanged. I am only adjusting the perfmtk profile. powersave+ is lower performance across the board but that's to be expected.

JUANIMAN, 2024-12-08:
maybe due to thermal throttling.

dbar24, 2024-12-08:
I tested the balanced profile while the phone was cold to the touch and got the same result, after switching to powersave profile the performance improved significantly. This is also an extremely thick phone and takes significantly longer to get warm than many other phones.

closed 2025-04-24 by JUANIMAN