JUANIMAN/PerfMTK #9: Keep booting into Fastboot for Mediatek Helio G99 Ultimate

aaronmurniadi, 2024-12-13:
After installing, the phone keeps booting into Fastboot. Need to delete/disable the module from recovery.

Phone: Infinix Note 40 4G (X6853)
Chipset: Mediatek Helio G99 Ultimate 
GPU: Mali-G57 MC2

Feel free to ask me for more details to help you debug 👍

JUANIMAN, 2024-12-13:
Does it only happen with the latest version?

aaronmurniadi, 2024-12-13:
Yes, currently this is happening to me on v7.5

JUANIMAN, 2024-12-22:
> Yes, currently this is happening to me on v7.5
> 
> Edit 1: Let me try v6.9 and I'll get back to you with the results. Edit 2: Unfortunately, it's also happening in v6.9

Can you try the new version?

aaronmurniadi, 2024-12-23:
All right, gimme a few minutes

aaronmurniadi, 2024-12-23:
> > Yes, currently this is happening to me on v7.5
> > Edit 1: Let me try v6.9 and I'll get back to you with the results. Edit 2: Unfortunately, it's also happening in v6.9
> 
> Can you try the new version?

Gimme a few mins

JUANIMAN, 2024-12-23:
Can you try this version with these options?
- [PerfMTKv7.7(preview).zip](https://github.com/user-attachments/files/18234093/PerfMTKv7.7.preview.zip)
- 2 - 2 - 2

aaronmurniadi, 2025-01-02:
Sorry I was on new years holiday, will check the new version after I got back from work 👍

JUANIMAN, 2025-01-02:
Don't worry, try the version I passed here.

Btw, happy new year 🎉

aaronmurniadi, 2025-01-05:
Hihi, with 2-2-2 option I can boot to the lockscreen briefly, then it crashed and reboots.

JUANIMAN, 2025-01-05:
> Hihi, with 2-2-2 option I can boot to the lockscreen briefly, then it crashed and reboots.

It's very strange, doesn't it happen to you with other modules?

aaronmurniadi, 2025-01-10:
> > Hihi, with 2-2-2 option I can boot to the lockscreen briefly, then it crashed and reboots.
> 
> It's very strange, doesn't it happen to you with other modules?

Nope