JUANIMAN/PerfMTK #11: Bootloop after install in Helio G85 (Android 14)

AaronSamuelBernal, 2024-12-21:
Hi, I have installed the module before on Android 13 and it worked correctly.
However, in Android 14 (crdroid), after installing and restarting the device, the system cannot boot and restarts.

JUANIMAN, 2024-12-21:
Maybe it's the custom rom, I'll try to make a more user-configurable version.

AaronSamuelBernal, 2024-12-22:
The Android 13 that I used before was also from Crdroid. Is it possible that it is due to an incompatibility with Android version 14?
If you need any more information that I can give you to help you, I am here.

JUANIMAN, 2024-12-22:
Thanks for the support, I have already been updating the module installation script, I think that most of the bootloop problems lie in the settings within post-fs-data.sh, the next update is very close

closed 2025-03-24 by JUANIMAN