JUANIMAN/PerfMTK #19: Bootloop on duchamp a15

handingslider, 2025-03-07:
Reboots on lockscreen and bootloops
Poco X6 Pro(duchamp)
All versions after 4.1

JUANIMAN, 2025-03-08:
Custom ROM? I also use Duchamp A15

JUANIMAN, 2025-03-08:
> Reboots on lockscreen and bootloops Poco X6 Pro(duchamp) All versions after 4.1

I made a small change, you might want to try it to see if it fixes the problem.

[PerfMTKv8.1.zip](https://github.com/user-attachments/files/19141362/PerfMTKv8.1.zip)

handingslider, 2025-03-08:
> Custom ROM? I also use Duchamp A15
What custom rom you using?

handingslider, 2025-03-08:
> > Reboots on lockscreen and bootloops Poco X6 Pro(duchamp) All versions after 4.1
> 
> I made a small change, you might want to try it to see if it fixes the problem.
> 
> [PerfMTKv8.1.zip](https://github.com/user-attachments/files/19141362/PerfMTKv8.1.zip)

Still same issue

JUANIMAN, 2025-03-09:
You can pass me the logs generated in /data/cache/recovery and /data/cache/magisk.log?

surf5726, 2025-03-22:
same issue when boots to the lockscreen, then system immediately crashed and reboot to the recovery mode.