JUANIMAN/PerfMTK #27: "install the latest version of module error"

rasik33, 2025-05-10:
In perfmtk manager when i choose one setting, i get error "install the latest version of the module" but i have installed the latest 
My phone is Redmi note 13 pro.

JUANIMAN, 2025-05-12:
Do you have the latest version of PerfMTK manager too?

Test the module from Termux, just to rule out if it is the module or the App.

rasik33, 2025-05-12:
Yes i have the installed the latest module and manager. In termux it works no problem, but when i reboot my phone it goes back to "balanced" in perfmtk manager. So i have to activate again in termux. Why is that?

rasik33, 2025-05-14:
I fixed by istalling old version module .

Citra-Standalone, 2025-09-08:
The toast was a bug, to fix it you need to give Root Access to PerfMTK Manager App.

rasik33, 2025-09-08:
Thats right, it was not about version or module . I just had to give Root access PerfMTK Manager App 👍

closed 2026-01-03 by JUANIMAN